openscopeproject/InteractiveHtmlBom #110: Export assembly drawing

Munken, 2019-09-10:
Hi

Is it possible to export the generated assembly figures to png, pdf or svg?
Either from the generated html page or as a commandline option.

Regards

qu1ck, 2019-09-10:
Are you looking for image with or without highlights of component groups? If with highlights, then separate images for all of them?
Have you looked at plotting in pcbnew, it can produce svg, or pdf of any layer. It's not exactly the same drawing as what this plugin generates but it might fit your needs.

Munken, 2019-09-11:
I've tried pcbnew.   
However, I'm missing a way to show the PTH without cluttering the drawing with traces.

Comparison:

InteractiveHtmlBom:  
[image]

PCBnew:  
[image]

qu1ck, 2019-09-11:
I see, so you want copper items but not tracks/zones.

I can offer 2 workarounds:
1. In pcbnew select all, then `rightclick -> select... -> filter selection...` and select only tracks and zones. Delete them. Plot to svg including appropriate layers. After plotting undo the deletion.
2. In InteractiveHtmlBom switch to single layer view (F or B buttons) and make the drawing as large as possible by dragging the delimiter bar. Take a screenshot using OS utilities (Win+Shift+S works on windows 10, I think cmd+shift+4 on macos).

It is not trivial to add download functionality in ibom itself. Currently the board is rendered on 4 separate canvases, I would have to rerender them on one combined canvas and then somehow trigger a download or "save as" action. I will look into this further and see if it's possible on file:/// url.

qu1ck, 2019-09-11:
This seems to work at least in chrome:
https://stackoverflow.com/a/30393357
I still have to test this in other browsers but it seems technically doable.
Not sure about UI for this...

Munken, 2019-09-16:
Thanks for the workarounds. I used option 2 for this.   
However, rendering directly as svg or pdf would be nice.

How about right click -> save as? That would be sufficient UI for me.

qu1ck, 2019-09-16:
Right click on image is already mapped to reset view.

Munken, 2019-10-23:
How about just adding a button?

qu1ck, 2019-10-23:
That's doable but I can only generate raster image (png, not svg or pdf).

Munken, 2019-10-23:
As long as the resolution can be adjusted, then this is fine.

closed 2019-11-25 by qu1ck

qu1ck, 2019-11-25:
You can export images up to 16000x16000 now.

commit bcc10634 (2019-11-25): Implement image export

Fixes #110