openscopeproject/InteractiveHtmlBom #114: localisation issue in units.py

BennoG, 2019-10-14:
In locations where decimal separator is a ',' kicad crashes with the message.
"in wxWidges with the comment setlocale called directly not using wxlocale.
now there is a mismatch between C/C== and Windows locale"

this can be fixed by commenting line 17 in units.py
   `#locale.setlocale(locale.LC_NUMERIC, '')`

Benno

qu1ck, 2019-10-14:
Can you post a screenshot of the message? Also add kicad version information and windows version.

BennoG, 2019-10-14:
Error message and kicad version below.

[image]

[image]

BennoG, 2019-10-14:
Updated to 5.1.4 it is working now

closed 2019-10-14 by BennoG