openscopeproject/InteractiveHtmlBom #118: Path "abbreviattion" at the text box

hildogjr, 2019-10-31:
I just realize that some times my files path is too big, example "/home/h/Documents/UNICAMP/Thesis/Boards/sbRIO Brazil/output-ligthTRIAC/ligthTRIAC.xml", and doesn't fit at the text box.

[image]

Some softwares workaround this making an abbreviation preview "/home/h/.../ligthTRIAC.xml" to be possible to see the maximum path string and prioritize the end (file name). When we click to edit, those software show the full path without abbreviation.

Is it possible (simple) to do with wxPython? I really don't know, because my [KiCad plugin project GUI](https://github.com/xesscorp/KiCost/) have this same behavior.

qu1ck, 2019-10-31:
I don't think this is doable with current wx.FilePickerCtrl. You will have to disable built in text control and add your own which you will update when file/folder is picked. But then it's not trivial to determine how to correctly ellipsize the text. Some platforms (looking at you GTK) are notoriously bad at predicting how much space an element or text will take so efficiently using non-monospaced width font is practically impossible.
If you figure out how to do it let me know :)

hildogjr, 2019-11-01:
I will search about. This also interests me in other piece of softwares.

qu1ck, 2019-11-01:
Sounds good. Feel free to reopen later but at this point this bug is not actionable for me.

closed 2019-11-01 by qu1ck