openscopeproject/InteractiveHtmlBom #119: Add the possibility to save and restore checkboxes

suzizecat, 2019-11-04:
Hi !

First of all, thanks for this wonderful plugin.
One nice feature would be to be able to save and restore the state of check boxes (to follow you progress while sourcing components, in example).

The absolute best would be to be able to save the whole html with the checkboxs state embedded by simply hitting `Ctrl-S` but anything less optimal (using an intermediate file ?) would be nice too.

Best regards,
Julien

qu1ck, 2019-11-05:
Use a decent browser (i.e. not edge) and it will save them for you.
Or are you talking about transferring checkboxes state between different computers?

suzizecat, 2019-11-05:
I’m using up-to-date firefox on Ubuntu and, unless I missed something, it doesn’t save anything.
Moreover it would be nice to be able to share through several computers, but it is not my main point.

qu1ck, 2019-11-05:
I just tried this on latest firefox on debian, it definitely saves the checkboxes and settings (like dark mode etc) across refreshes and restarts of the browser.
Couple questions:
1. Is html bom openeded from local filesystem (i.e. url starts with file:///) or is it hosted somewhere, even if it's http://localhost ?
2. Open storage viewer in firefox (shift+f9) and go to localstorage, select opened file. Play with the settings and checkboxes. Do you see values for your file change? Keys are prefixed with "KiCad_HTML_BOM__your_board_name__your_board_rev". When you restart browser, open bom file and open storage viewer again, are the values still there?

easyw, 2019-11-06:
@qu1ck 
if you store your settings in localstorage, those will be deleted if the browser is set to "Delete cookies and site data when Firefox Developer Edition is closed"
This is a quite common setting for privacy ...
Would it be possible to add an option to save the settings in a '.ini' file beside the html file itself?

suzizecat, 2019-11-06:
@qu1ck HTML BOM is oppened from the local filesystem (`file://<stuff>` URL).
I will fiddle around with Firefox parameters later today.

suzizecat, 2019-11-06:
@qu1ck For whatever reason, it seems to work fine today.
I still think that it would be interesting to save those data in some actual file, but I'm fine with the current state.

qu1ck, 2019-11-06:
I'll keep export/import of settings and checkboxes as a wishlist item.

commit d40ee98b (2019-11-25): Implement settings export/import

Fixes #119
closed 2019-11-25 by qu1ck