openscopeproject/InteractiveHtmlBom #121: Bézier curves in EdgeCuts do not show properly in the board drawings

Muny, 2019-11-13:
I imported a DXF drawing for my board outline onto EdgeCuts, and when viewing the board drawings in the exported BOM, the Bézier curves in the edge cuts do not appear. See the images below.

[image]

And in KiCAD:
[image]

Here is a sample project that has one of these curves which you can use to reproduce the issue:
https://f.pcbx.us/ge

Thanks! This is a very awesome tool, it has become part of my workflow for working with KiCAD.

qu1ck, 2019-11-13:
Yes, Bezier curves are only in nightlies and not supported yet by plugin. I'll implement them soon.

qu1ck, 2019-11-14:
Fixed in 8587d54

closed 2019-11-14 by qu1ck

Muny, 2019-11-14:
Just tried and it works. Very nice job! Thanks.

qu1ck, 2019-11-14:
I noticed on your screenshots that drill holes of the edge connector pads in kicad are offset but not in ibom render. Is that just because board was modified after bom was generated or is that a bug too?

Muny, 2019-11-14:
Oh, I hadn't noticed that. They are offset in the design, and haven't been changed. The field I used to do this is "Shape Offset".

qu1ck, 2019-11-14:
Fixed in a143bd9