openscopeproject/InteractiveHtmlBom #124: Make default column sort order in html configurable

qu1ck, 2019-11-26:


Halbelunge, 2020-06-04:
Is there any possibilty to sort the columns at the moment?

qu1ck, 2020-06-04:
Yes, you can sort columns by clicking on the header. But it will reset on page reload and you can not specify it at bom generation time. This bug is about fixing the latter.

Halbelunge, 2020-06-04:
Maybe "sort" was the wrong word. I do not want to sort the rows content regarding to one column, I want to sort the order of the columns from left to right. So as an example change "value" with "quantity".

I found a way to modify the html afterwards, but would be nice to do this in the GUI. I think to have the extra fields at the end, not in the middle would be better :) (personal oppinion)

qu1ck, 2020-06-04:
Not implemented yet, #123 covers that.

Halbelunge, 2020-06-05:
Thank you :)

For the moment I know how to reorder manually :)

kepp going with this great project!