openscopeproject/InteractiveHtmlBom #125: Feature requestJS callouts extentability

tablatronix, 2019-11-29:
Some kind of JS event callout hooks to extend functionality, eg. 

I am playing around with something like this, could also be tied into things like inventory control and picking etc.

https://youtu.be/oZNod_Mhfro

qu1ck, 2019-11-29:
Can you expand on what you mean by JS callouts? Maybe an example of integration with outside system?

tablatronix, 2019-11-29:
That yt video is an example, and the desc is pretty large.

Just events for current selected item etc.

I just hacked in an xhr for example to pass data to a host I control in this example...

```JS
function createRowHighlightHandler(rowid, refs) {
...
     URL = "http://192.168.1.246/?ledindex="+rowid.replace("bomrow","");
    let xhr = new XMLHttpRequest();
    xhr.open("GET", URL,true);
    xhr.send();
```

qu1ck, 2019-11-29:
I see. Well these are hardly JS callouts, more like http callbacks but it's semantics.
I will think about good way to implement it. What events other than highlight do you think would be useful to have callbacks for?

tablatronix, 2019-11-29:
sorry , I meant callouts in JS, not actual 'JS callouts' is that a thing?

functionally to callout to a users control function or trigger an event for customization or action.

qu1ck, 2019-11-29:
I'm not an expert but probably in pure JS apps it's a thing, not in web since browsers sandbox everything.

tablatronix, 2019-11-29:
ok well to be clear the xhr call has nothing to do with this.

Might be able to just get away with event listeners, but it depends on the scope of how to allow including user code assets. And what data we need to pass to the function etc

qu1ck, 2019-11-29:
I agree, it's better to make it generic and not limited to http requests. 
I think I'll add ability to inject user js code at bom generation time and that code will be able to add listeners to some events like highlight change, checkbox state change, etc. Then you can do whatever you want in your code.

tablatronix, 2020-04-26:
Have you thought about this anymore?

qu1ck, 2020-04-27:
Yes, I have a plan to implement the injectable user js code with listeners as described above and also injectable css and html headers/footers. Just have to find time to implement it.

qu1ck, 2020-05-01:
This is now supported. In `InteractiveHtmlBom/web/user-file-examples` there are some examples you can copy over into `InteractiveHtmlBom/web` to check how they work and edit as you like.

I will add more info to the wiki soon.

commit c71cebc6 (2020-05-01): Add custom js css and header/footer support

Fixes #125
Fixes #72
closed 2020-05-01 by qu1ck