openscopeproject/InteractiveHtmlBom #126: Variety of JS errors

C44Supra, 2019-12-10:
I'm using the most recent version of this plugin and I'm having quite a few issues.

I'm using all default options. No components are virtual and all have proper names etc. Nothing out of the ordinary. I'm running the plugin from within KiCAD. Over in the browser (any browser) I get not output past the "ref lookup" and "filter" fields:
[image]

In the element inspector / console I get the following output 
```
Uncaught TypeError: Cannot read property 'includes' of undefined
    at populateMetadata (ibom.html:2744)
    at window.onload (ibom.html:3004)
```

I can dig into the html file itself and on line 2744 I see
`if (pcbdata.bom.skipped.includes(i)) continue;`

To test I removed that line of code and reloaded the page. I don't see any definitions of the skipped attribute anywhere in the document by the way, but its sub attribute includes is queried a number of times for length. Those queries all result in more JS errors:

```
ibom.html:2925 Uncaught TypeError: Cannot read property 'length' of undefined
    at updateCheckboxStats (ibom.html:2925)
    at prepCheckboxes (ibom.html:2919)
    at window.onload (ibom.html:3011)
```
If I now click any of the layout buttons ("BOM only" in this case), I suddenly see the BOM list. But, when I mouseover any one of the items in the list I now get:

```
Uncaught TypeError: ctx.translate is not iterable (cannot read property Symbol(Symbol.iterator))
    at drawModule (ibom.html:1534)
    at drawModules (ibom.html:1584)
    at drawHighlightsOnLayer (ibom.html:1675)
    at drawHighlights (ibom.html:1685)
    at HTMLTableRowElement.<anonymous> (ibom.html:2355)
```
I've exported the same BOM a number of times, with and without components on the blacklist but none of it makes a difference.

qu1ck, 2019-12-10:
It sounds like you have partially updated the plugin or something like
that. Try removing it completely and installing from scratch.
If that doesn't work then try generating bom from kicad demo or any other
project file and share it with me so that i can take a closer look at
exceptions.

C44Supra, 2019-12-10:
Hah, that's odd. I did upgrade from an earlier version but I removed the entirety of the old folder then too. Only thing I *think* may be different this time around is that I actually shut down KiCAD before removal and re-install and only then started it back up again. I guess we can close this issue then.

closed 2019-12-10 by C44Supra