openscopeproject/InteractiveHtmlBom #128: Group and propagate similar components unless extra fields are different

whatnick, 2019-12-21:
Would like to group all Capacitors of 0603 footprint 27nF together unless they have DNP or different manufacturer part number. Is this possible ? I currently use #manf as per KiCost plugin to set manufacturer part number. However iBOM currently splits the parts for which this custom field is set. In large boards being explicit with manufacturer part numbers may be more desirable.

[image]

qu1ck, 2019-12-21:
I don't quite understand the issue, you seem to be contradicting yourself.

> Would like to group all Capacitors of 0603 footprint 27nF together unless they have DNP or different manufacturer part number.

Isn't this exactly what is happening? You have 2 rows because you have a group of caps with manf# not set and one with manf# set. Field that is not set is considered empty string and it's different from field that is set.

> However iBOM currently splits the parts for which this custom field is set.

And that is not what you expect it to do? Here is the contradiction.

whatnick, 2019-12-21:
I would like a toggle to infer that all parts with the same footprint and
value have the same #MPN. This behaviour speeds up setting MPN (sort of
implicitly) and was the default behaviour in KiCost which I was using
previously as my default BOM module. This behaviour may not suit all
boards, but would be great for me to have.

Please see KiCost part grouping behaviour details here :
https://kicost.readthedocs.io/en/latest/usage.html#part-grouping

Much better documented there compared to my brief in the issue. It would be
great to have and would reduce effort. My boards are small so far and I am
happy to pay the extra effort.

qu1ck, 2019-12-21:
Ok, I understand now.
I don't really agree that this is the right approach. KiCost has generally different goals and this may work for their purposes.
I think that having empty field can and should be interpreted as "any part number of same value and footprint" and should be separated from non-empty field where specific part is required (maybe because having x7r there is critical but not in other parts, just as an example).

I'm especially not inclined to implement this in ibom because your use case is very simple to solve from KiCad itself. Just open field editor in eeschema, it shows bom-like grid where you can assign same part number to all of your 27nf caps in one copy paste operation.

whatnick, 2019-12-22:
Ah yes better get that KiCAD 5.1 upgrade in place. Thanks for the push.
Feel free to close this, will leave it in place for future users.

closed 2019-12-22 by whatnick