openscopeproject/InteractiveHtmlBom #135: Post edit html

WhiteChairFromIkea, 2020-02-03:
Hi,
It is useful to manually edit BOM table (only strings, part values) once html file is generated. For instance: I would like to run second assembly batch of the same board, but I need to change all capacitors from 6.3V rating to 10V, because I do not have 6.3V version in stock at the moment, but have plenty of 10V. Previously, I opened html with in notepad, and replaced all "6.3V" strings to "10V", and this change was sufficient to update BOM table. It is impossible in recent version of the plugin, because pcb data is compressed, not plain text as in previous releases, I mean data is "encrypted" inside "var pcbdata = JSON.parse(LZString.decompressFromBase64()". Is it possibe to edit BOM table easily, without opening KiCad? Thank you for extremely useful work! ;)

qu1ck, 2020-02-03:
You can decrypt, edit and re-encrypt the pcbdata using this page
https://jsfiddle.net/4b1dmk89/5/

Does this work for you?

WhiteChairFromIkea, 2020-02-04:
Thank you, this solved my issue, case closed :)

WhiteChairFromIkea, 2020-02-04:
> You can decrypt, edit and re-encrypt the pcbdata using this page
> https://jsfiddle.net/4b1dmk89/5/
> 
> Does this work for you?

Thank you, this solved my issue, case closed :)

qu1ck, 2020-02-04:
Great. In the future I will also add option to disable compression, there are some cases where it is better left off.

closed 2020-02-04 by qu1ck

iharshadec, 2021-01-25:
I am not sure if option to disable compression is added on new releases or not. But just encounter situation exactly same and need to change things.
For now I am going to try decrypt and encrypt thing.