openscopeproject/InteractiveHtmlBom #136: Extra fields not updated

markus-k, 2020-02-27:
When updating extra fields in Eeschema (like DNP, MfrNo etc.), then updating the PCB from schematic in PCBnew, and then regenerating the HTML BOM, the previously updated fields are not showing the correct value. I.e. when you mark a component as DNP later on, it won't be marked as DNP in the BOM.

I'm not sure if this is a problem with InteractiveHtmlBom or KiCad itself, but I didn't notice such problems outside of InteractiveHtmlBom before.

qu1ck, 2020-02-27:
It's not enough to simply update fields in schematic, you need to regenerate netlist file or xml file. Ibom doesn't read schematic directly.

markus-k, 2020-02-27:
Okay, using the netlist does the trick. But it doesn't work with "Update PCB from schematic", it seems to be a KiCad problem then?

qu1ck, 2020-02-27:
No. Since v5 kicad doesn't use netlist as intermediate file when you do direct update from pcbnew so the netlist file doesn't get updated. Manual eeschema -> netlist -> pcbnew workflow is still supported but not for long.
You just need to remember that data sources for ibom are .kicad_pcb and .net/.xml files. If you want field data from eeschema you need to manually regenerate .net/xml.
If eeschema api makes it into v6 I'll make this more seamless but for now that's how it is.

markus-k, 2020-02-27:
Ah alright, so that was a misunderstanding from my side. Thanks for clearing that up!

closed 2020-02-27 by markus-k