openscopeproject/InteractiveHtmlBom #138: [wish] Save, save as and load the current board.

viejolobo, 2020-03-20:
I would like to use the tool in small series production. To do this, I have to be able to save the current status of the circuit board and load it again later.

I want to specify a path where I want to save the file.

qu1ck, 2020-03-20:
[image]

How do existing options not work for you?

viejolobo, 2020-03-24:
No, I mean while working with the ibom...
[image]

qu1ck, 2020-03-24:
So you want export button to open a "Save as" dialog instead of just putting it in browser downloads?
Not sure what exactly you meant by state in the first message. Checkboxes?

viejolobo, 2020-03-24:
I want to use iBom in the production of small series. For this I want to write a file with a serial number after assembling the first PCB. Then populate the second PCB etc. At the end I have n files where all components are populated and all check boxes are checked.
Saving as in the browser only saves the bare HTML ...

qu1ck, 2020-03-24:
Right, so you want to store the state of checkboxes, that's what the export button is for. I'll look into how to make a save as dialog appear.

Or do you want a button that will save html + checkbox state in that html?

qu1ck, 2020-06-24:
Closing as not feasible. Both saving html and download "save as" dialog is controlled by browser.

closed 2020-06-24 by qu1ck