openscopeproject/InteractiveHtmlBom #148: Data Export Feature?

mattandwhatnot, 2020-04-30:
It would be cool if you could export selected column data right from this as a .CSV. This could be done out of KiCAD proper but for the workflow at our shop it would be more streamlined if it could be pulled from this. 
Main use case would be the part/location/XY for our pick'n'place equipment. (We run a mixed group of machines with different CAD formats so we feed everything from a root .CSV. But maybe other shops differ, though its seems too verbose to support export for all the different line setups. Who else uses the 2000s era Hitachi .MSE format, right?)

qu1ck, 2020-04-30:
There is export to tab separated values by clicking copy button 

[image]

you can paste that into spreadsheet software or notepad and replace tabs with commas.

Implementing export in every possible format is not viable.

qu1ck, 2020-04-30:
Also if you want part location data in your export it will have to be custom coded. One way is to write your own plugin to pull from kicad, another is add custom js to ibom once that is supported (see #125)

closed 2020-04-30 by qu1ck

mattandwhatnot, 2020-05-01:
Of course export in all the different formats is untenable. Of course the copy function allows for Xcel manipulation of the part columns but I was talking about location data and the potential for avoiding that Xcel manipulation and path from kiCAD. 

Maybe I was looking at it too specifically for integrating with our shop's specific arrangement. 

So fair enough, custom JS is the way to go

qu1ck, 2020-05-02:
Yep, custom js will allow you to do that and more.
This is already implemented, see [customization](https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Customization).