openscopeproject/InteractiveHtmlBom #152: Problem

amagnard, 2020-05-03:
this is the message i got File ”/private/var/folders/py/y5mcfbb55yjd62zr7gp7wyhc0000gp/T/AppTranslocation/66A7254F-7EFE-4965-B9B9-9FC1BA76607B/d/kicad.app/Contents/SharedSupport/plugins/InteractiveHtmlBom/generate_interactive_bom.py”, line 7, in <module>
    import wx
ImportError: No module named wx
Any ideas
regards

drojf, 2020-05-03:
How are you running the script? Is it through the command line, or is it from kicad by pressing the button? it looks like it's being run using your system's python rather than Kicad's python, hence the missing module.

Also just to double check, you're on MacOS right?

qu1ck, 2020-05-03:
Yeah this issue would be if plugin is run from command line using system python. Use kicad's python instead.
If you are trying to generate bom for EasyEDA file then install wxpython using pip.

amagnard, 2020-05-04:
Hello
I have run your command line in eeschema your script and that's where I got the problem.
can you tell me what commands to use with the kicad shell python console?
I am not very comfortable with python compared to language C. I join 2 png to show you the error. Thanks for your help.

magnard48@gmail.com

amagnard, 2020-05-04:
Hello 

In the path /Library/Application Support/kicad/scripting/plugins it is not working.

I installed the plugin in /Users/alainmagnard27/Library/Preferences/kicad/scripting/plugins and it’s work now.
regards

Magnard gmail
magnard48@gmail.com

qu1ck, 2020-05-04:
Yeah, the second path is listed in [instructions](https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Installation). Note the `~` before path that denotes the home directory.

closed 2020-05-04 by qu1ck