openscopeproject/InteractiveHtmlBom #154: ~Inverted~ names on silkscreen not supported

ColinBathe, 2020-05-11:
If you have text on the silkscreen layer that uses ~ to add a bar on the top, then this isn't supported on BOM html output display.

i.e.
If you have text on the silkscreen layer which contains ~Reset~ then this will appear within Kicad as Reset with a bar over the top. However, within the BOM html output, it appears as ~Reset~

Yeah, ok, this isn't a high priority to fix.

ColinBathe, 2020-05-11:
[image]
[image]

qu1ck, 2020-05-11:
I have a fix ready but it's pending resolution on https://gitlab.com/kicad/code/kicad/-/issues/4396
Text rendering logic in kicad may change slightly and I'll follow suit.

commit d14cdc93 (2020-05-12): Text rendering fixes

* process overbars
* approximate italics behavior better

Fixes #154
closed 2020-05-12 by qu1ck