openscopeproject/InteractiveHtmlBom #155: Plugin does not show up on external plugin list, macOS Catalina, KiCad 5.1.6-0-10_14

KT819GM, 2020-05-18:
Plugin is placed into: `~/Library/Preferences/kicad/scripting/plugins/`. Action plugins shows up when I put them there. Any guidance to solve problem?

p.s. insane useful plugin
Thank you!

qu1ck, 2020-05-18:
How exactly did you install the plugin? Plugin should be in a directory and `__init__.py` should be in directory immediately inside, i.e. you can not wrap plugin into another directory.

Open scripting console in pcbnew and type in following one line at a time:
```
import pcbnew
print(pcbnew.NOT_LOADED_WIZARDS)
print(pcbnew.FULL_BACK_TRACE)
```

KT819GM, 2020-05-18:
Here is trace from kicad console. Do I have problems with python installation?

```bash
import pcbnew
print(pcbnew.NOT_LOADED_WIZARDS)
Skip subdir /Users/user/Library/Preferences/kicad/scripting/plugins
Skip subdir /Users/user/Library/Preferences/kicad/scripting/plugins/screenshots
/Users/user/Library/Preferences/kicad/scripting/plugins/InteractiveHtmlBom
print(pcbnew.FULL_BACK_TRACE)
Traceback (most recent call last):
  File "/private/var/folders/lj/ww3x_4t1553crdd78b71sjvw0000gn/T/AppTranslocation/07E5E86F-03A4-4347-B021-416AE503C951/d/kicad.app/Contents/Frameworks/python/site-packages/pcbnew.py", line 2966, in LoadPluginModule
    mod = __import__( ModuleName, locals(), globals() )
  File "/Users/user/Library/Preferences/kicad/scripting/plugins/InteractiveHtmlBom/__init__.py", line 47, in <module>
    from .ecad.kicad import InteractiveHtmlBomPlugin
  File "/Users/user/Library/Preferences/kicad/scripting/plugins/InteractiveHtmlBom/ecad/kicad.py", line 8, in <module>
    from ..core import ibom
  File "/Users/user/Library/Preferences/kicad/scripting/plugins/InteractiveHtmlBom/core/ibom.py", line 13, in <module>
    from . import units
  File "/Users/user/Library/Preferences/kicad/scripting/plugins/InteractiveHtmlBom/core/units.py", line 18, in <module>
    locale.setlocale(locale.LC_NUMERIC, '')
  File "/private/var/folders/lj/ww3x_4t1553crdd78b71sjvw0000gn/T/AppTranslocation/07E5E86F-03A4-4347-B021-416AE503C951/d/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7/locale.py", line 581, in setlocale
    return _setlocale(category, locale)
Error: unsupported locale setting
```
Thank you

qu1ck, 2020-05-18:
No, kicad packages it's own python on OSX. Plugin is installed correctly.

What does this print in scripting console
```python
import locale
locale.setlocale(locale.LC_NUMERIC)
locale.setlocale(locale.LC_NUMERIC, '')
```

Also try switching kicad to english or another language and back. Reload kicad after switching, check if plugin shows up.

KT819GM, 2020-05-18:
After switching language from `default` to `English', restarting Kicad, plugin does not show up. Console log of locale:
```bash
import locale
locale.setlocale(locale.LC_NUMERIC)
'en_GB'
locale.setlocale(locale.LC_NUMERIC, '')
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/private/var/folders/lj/ww3x_4t1553crdd78b71sjvw0000gn/T/AppTranslocation/037DC8C7-6D28-4F8F-8D35-7119DBC4FDDE/d/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7/locale.py", line 581, in setlocale
    return _setlocale(category, locale)
Error: unsupported locale setting
```

qu1ck, 2020-05-18:
I just pushed a possible fix, try downloading latest revision.
https://github.com/openscopeproject/InteractiveHtmlBom/archive/master.zip
Make sure to completely remove old plugin folder before installing new one.

KT819GM, 2020-05-18:
Works perfectly, thank you!

closed 2020-05-18 by qu1ck

MichMich, 2020-07-03:
I had the same issue. Your possible fix made it work for me.