openscopeproject/InteractiveHtmlBom #156: Double-click on checkbox columns headers doesn't work when "darken when checked" option is enabled

ChernyshevDS, 2020-06-15:
When double-clicked on header, table column with checkboxes (either Sourced or Placed) is supposed to check or uncheck all contained checkboxes. Unfortunately, it doesn't work when this column is selected in `darken when checked` option menu. I tracked this error to `checkboxSetUnsetAllHandler` - it calls `checkbox.onchange()` handler directly, without creating event args object, which in turn causes `createCheckboxChangeHandler` to fail when trying to use event arguments to find parent elements. Sorry, but my JavaScript-fu is not strong enough to fix it myself :(.
As a temporary workaround, the user may set `darken when checked` option to `None` before clicking on column headers.

qu1ck, 2020-06-15:
Thanks for the report and digging into the cause.

closed 2020-06-15 by qu1ck

ChernyshevDS, 2020-06-15:
Thanks for the olympic fix speed :)

commit c87a8bb8 (2020-06-15): Fix double click not working with "darken when checked"

Fixes #156