openscopeproject/InteractiveHtmlBom #160: Create .po file for volunteer to translate this plugin

scarrrr316, 2020-06-21:
same as title , make your plugin multilingual !!

you can gettext of python create the .pot file and use poedit (or other po file edit software )to  convert it to .po file ,than volunteer can translate it to other language

you can find more information by searching "python i18n"

and you can also create team on [Transifex](https://www.transifex.com/) and share link to others people to help you for translation

（sorry for my english haha）

scarrrr316, 2020-06-30:
I tried to make it but i just know a little python knowledge,emmmmm

scarrrr316, 2020-09-24:
emmm

scarrrr316, 2020-12-08:
lol

qu1ck, 2020-12-08:
It's not enough to just create a po file and have some translations. Those translations need to work out of the box and detect kicad's and/or system language settings. 
This is low on my list of priorities which is why there is no progress so far.

Unless you want to contribute working code please refrain from meaningless comments.