openscopeproject/InteractiveHtmlBom #161: Should put fitted and DNP parts on separate lines.

dfnr2, 2020-06-23:
To reproduce:
1) Create a design with several identical components (e.g., 10K resistors)
2) in one-per-line mode, select a subset of these to be fitted/populated.
3) switch to multiple-per-line mode.  All the designators are on the same line.  The fitted and not-fitted parts should be on separate lines.

qu1ck, 2020-06-23:
I'm assuming by one-per-line mode and multiple-per-line mode you mean grouped and ungrouped view in the generated bom.

If so, this behavior is by design. You should mark what is to be fitted and what shouldn't be populated before you generate the bom, either in schematic using extra fields or in pcbnew marking footprints as virtual.

Checkboxes do not affect bom grouping once it is already generated, they are there only to keep track of progress for various stages of making a finished board.

qu1ck, 2020-06-24:
Unless you provide further information I'll close this issue within a week.

closed 2020-07-03 by qu1ck