openscopeproject/InteractiveHtmlBom #163: Install doesn't appear in plugins menu

judge2005, 2020-07-03:
I need some help for how to trouble-shoot this (like is there a log file I can look at to see what KiCAD is doing?), as this is all I can tell you ATM. I tried refreshing plugins, but it still didn't show. I am running KiCAD 5.1.3 on Windows. I have one other plugin installed which appears fine. Here is where it is installed:

[image]

qu1ck, 2020-07-03:
Is there `__init__.py` file inside that folder? If not you messed up somewhere during unpacking/copying.

There is no log file but you can check for init errors by running this in scripting console (copy one line at a time):
```python
import pcbnew
print(pcbnew.NOT_LOADED_WIZARDS)
print(pcbnew.FULL_BACK_TRACE)
```

qu1ck, 2020-07-03:
Also don't use 5.1.3 it was never released properly because it had significant bugs. Update to 5.1.6.

judge2005, 2020-07-03:
Will do. This is what I get when I run those commands:

Py 0.9.8
Python 2.7.16 (default, Aug 15 2019, 17:51:40)  [GCC 9.2.0 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
Startup script executed: C:\Users\mpand\AppData\Roaming\kicad\PyShell_pcbnew_startup.py
import pcbnew
print (pcbnew.NOT_LOADED_WIZARDS)
C:\Program Files\KiCad\share\kicad\scripting\plugins\FPC_(SMD_type)_footprintwizard.py
Skip subdir C:\Users\mpand\AppData\Roaming\kicad\scripting\plugins
C:\Users\mpand\AppData\Roaming\kicad\scripting\plugins\InteractiveHtmlBom
print(pcbnew.FULL_BACK_TRACE)
Traceback (most recent call last):
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/pcbnew.py", line 2946, in LoadPluginModule
    mod = __import__( ModuleName, locals(), globals() )
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/FPC_(SMD_type)_footprintwizard.py", line 156, in <module>
    FPC_FootprintWizard().register()
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/HelpfulFootprintWizardPlugin.py", line 242, in __init__
    FootprintWizardParameterManager.__init__(self)
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/HelpfulFootprintWizardPlugin.py", line 32, in __init__
    self.parameters = {}
AttributeError: can't set attribute
Traceback (most recent call last):
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/pcbnew.py", line 2946, in LoadPluginModule
    mod = __import__( ModuleName, locals(), globals() )
  File "C:\Users\mpand\AppData\Roaming\kicad\scripting\plugins/InteractiveHtmlBom/__init__.py", line 47, in <module>
    from .ecad.kicad import InteractiveHtmlBomPlugin
ImportError: No module named kicad

qu1ck, 2020-07-03:
That error points to `ecad/kicad.py` file missing. Looks like you didn't unpack/copy all the files of the plugin or maybe directory structure was not preserved.

judge2005, 2020-07-04:
Yep. My bad. I had some issues unpacking it and thought I had fixed it. I re-downloaded the ZIP and unpack it properly and it works. Thanks for helping me.

qu1ck, 2020-07-04:
No problem, enjoy your fancy BOMs :)

closed 2020-07-04 by qu1ck