openscopeproject/InteractiveHtmlBom #165: Workaround for crash on nightly build 5.99

eymre, 2020-07-07:
Just wanted to add a quick workaround for anyone who is trying to use this plugin on the nightly build and see the crash with the message below. Sorry couldn't find a better place to post this, please feel free to close/remove this issue.

```
[Window Title]
Exception on python action plugin code

[Main Instruction]
Traceback (most recent call last):

[Content]
  File "C:\Users\Emery\AppData\Roaming\kicad\5.99\scripting\plugins/InteractiveHtmlBom-master/InteractiveHtmlBom/ecad/kicad.py", line 523, in Run
    ibom.run_with_dialog(parser, config, logger)
  File "C:\Users\Emery\AppData\Roaming\kicad\5.99\scripting\plugins/InteractiveHtmlBom-master/InteractiveHtmlBom/core/ibom.py", line 328, in run_with_dialog
    main(parser, config, logger)
  File "C:\Users\Emery\AppData\Roaming\kicad\5.99\scripting\plugins/InteractiveHtmlBom-master/InteractiveHtmlBom/core/ibom.py", line 293, in main
    pcbdata, components = parser.parse()
  File "C:\Users\Emery\AppData\Roaming\kicad\5.99\scripting\plugins/InteractiveHtmlBom-master/InteractiveHtmlBom/ecad/kicad.py", line 469, in parse
    drawings, pcbnew.F_SilkS, pcbnew.B_SilkS),
  File "C:\Users\Emery\AppData\Roaming\kicad\5.99\scripting\plugins/InteractiveHtmlBom-master/InteractiveHtmlBom/ecad/kicad.py", line 201, in parse_drawings_on_layers
    drawing = self.parse_drawing(d[1])
  File "C:\Users\Emery\AppData\Roaming\kicad\5.99\scripting\plugins/InteractiveHtmlBom-master/InteractiveHtmlBom/ecad/kicad.py", line 168, in parse_drawing
    return self.parse_text(d)
  File "C:\Users\Emery\AppData\Roaming\kicad\5.99\scripting\plugins/InteractiveHtmlBom-master/InteractiveHtmlBom/ecad/kicad.py", line 159, in parse_text
    "thickness": d.GetThickness() * 1e-6,
AttributeError: 'TEXTE_PCB' object has no attribute 'GetThickness'

[OK]
```

The following changes in two places inside the %APPDATA%\kicad\5.99\scripting\plugins\InteractiveHtmlBom-master\InteractiveHtmlBom\ecad\kicad.py file worked for me (note that the plugins path for nightly is somehow different than standard installation).

118: Find:
```point = outline.Point(point_index)```
Replace with:
```point = outline.CPoint(point_index)```

159: Find:
```"thickness": d.GetThickness() * 1e-6,```
Replace with:
```"thickness":0.1,```

qu1ck, 2020-07-07:
KiCad nightly changes relatively fast, when using it one should use "nightly" version of the plugin as well, instead of release version.
In this case both issues are fixed already in the master branch, you can download it by clicking green button on github project homepage.
But please do report issues you notice with nightly if they are not yet fixed in the latest version of the plugin.

closed 2020-07-07 by qu1ck

eymre, 2020-07-07:
Perfect. Thanks for the reply.