openscopeproject/InteractiveHtmlBom #166: Crash on GUI, but works in comand line

davidelta, 2020-07-07:
Hi,
I discover this plugin and I tried.
My specs are:

- Windows 10 Pro version 1909 (18363.900)
- Kicad 5.1.4-1 release build (now I see too: platform Windows 8(build 9200) 64 bit in kicad -> about)

I installed the plugin in C:\Users\<my user>\AppData\Roaming\kicad\scripting\plugins\InteractiveHtmlBom
I can see the icon on pcbnew program. When I push the button, crash and says:

[image]

I was supposed the problem in the blank spaces in the route of files. I make a copy of the project and rename all files to remove the blank space. Same behavior.
I try with command line, and works!
[image]

What's possible wrong?

Thanks.

qu1ck, 2020-07-07:
This is caused by malformed xml file in your project directory. If you remove it GUI will work fine as well.

Plugin should handle this properly instead of crashing though, that's a bug.

davidelta, 2020-07-09:
Hi,
you are right. The xml file in the project was cut.

[image]

If i rename the file, the GUI appears. But my PCB is empty!
You know how to remake this file? This file is made by your extension or by kicad?

Thanks.

qu1ck, 2020-07-09:
The xml file is generated by eeschema as a byproduct of it's own bom generation method (`Tools` -> `Generate Bill Of Materials...`).
It's much easier to generate netlist file and delete the xml, netlist file has the same information.

I don't understand this part
> But my PCB is empty!

It wasn't empty before you deleted xml file? Is it empty in pcbnew or in generated ibom page? Existence of xml file is not related to pcb at all.

davidelta, 2020-07-10:
When I renamed the XML file, my pcb on pcbnew was empty. Today, I repeat the same operation, but the pcb was showed. I think was a fault of internet connexion, because I'm working through a VPN. Maybe slow connection.

With the XML renamed, i can push the GUI button and the program show me the window to configure the output. And the output is generated correctly. That's so beautiful.

I discover why the XML file was corrupted.
When I make the netlist in eeschema, it makes a file with extension .NET instantly. The file is 42Kb of size.
But when I make the BOM in eeschema with bom2grouped_csv script, it makes a XML file of 1Kb instantly, but this file growns about 1Kb every 8 seconds. The final file is 48Kb of size. It take a few minutes to generate the XML file.
Because the program freezes and I don't see any file with BOM or CSV extension (i thinked the script makes BOM with this extensions), i believed the BOM was not generated.

The problem was solved. Can I close this issue or you prefer leave open to make changes in your project?

Thanks a lot.

qu1ck, 2020-07-10:
Yes, xml file is finicky and depends on a lot of variables, that's why I recommend to use .net file if you want extra field information. (Neither is required for basic operation of the plugin, only one or the other is needed for extra fields).

> The problem was solved. Can I close this issue or you prefer leave open to make changes in your project?

Leave it for now, I'll close it once I fix the bug related to malformed files.

commit e0c3b877 (2020-07-11): Catch exceptions on parsing net/xml files

Fixes #166
closed 2020-07-11 by qu1ck