openscopeproject/InteractiveHtmlBom #167: Make "Footprint" and "Value" fields optional

cebrax, 2020-07-09:
When creating BOMs with the extra field named for example "Details", I have no need for "Footprint" or "Value" fields..

For example my "Details" field will include "Capacitor Ceramic Chip SMD 100nF 25V 0805 5%".. 

It would be great to disable "Footprint" and "Value" fields..

Also, "Footprint" field reveals footprint library item name which may be sensitive or ugly such as "IPC-DO-214AB(SMC)"

Pun-e, 2020-11-06:
I came here for the same issue

We have a proper itemised inventory with our own codes.

[image]

Being forced to see this completely useless column is an eyesore on an otherwise _gorgeous_ plugin

qu1ck, 2020-11-06:
While I plan to implement this as a plugin option, you can do this already using some custom js.
Just create a `user.js` file in `web` directory of the plugin with the following contents:

```js
var customRuleIndexes = [];

function hideColumn(column) {
    customRuleIndexes.push(document.styleSheets[0].insertRule(
        `td:nth-child(${column}),th:nth-child(${column}) { display: none }`,
        document.styleSheets[0].cssRules.length
    ));
}

EventHandler.registerCallback(IBOM_EVENT_TYPES.BOM_BODY_CHANGE_EVENT, (event) => {
    while (customRuleIndexes.length > 0) {
        document.styleSheets[0].deleteRule(customRuleIndexes.pop());
    }
    var numCheckboxes = event.args.checkboxes.length;
    hideColumn(numCheckboxes + 3); // Value
    hideColumn(numCheckboxes + 4); // Footprint
});
```

This will hide Value and Footprint columns. Tweak the indexes to account for extra fields or hide other columns as needed.

cebrax, 2020-11-06:
Hey, thank you for this! This has been a great option for me..

And also thank you a ton for this GREAT plugin! By far the best plugin for KiCad!

Pun-e, 2020-11-06:
Thanks for the quick solution!

I've used VS Code to create a new javascript file, confirmed it's in the right folder, but even after restarting KiCad a few times, my interactive BOM still looks the same?

Sorry I'm so inept with js, literally never used it before 🙏

ArnieO, 2021-02-15:
What am I doing wrong when I cannot make the user.js file work as described?

qu1ck, 2021-02-15:
I kind of misplaced my crystal ball so help me out here. Do you see any effect at all when you add user.js file and regenerate bom? If you don't see any effect then try putting `alert('test')` in user.js instead and then regenerate bom. Upon loading the page it should pop a window with message "test". If that doesn't happen then you are not putting the file in the right directory. Or windows is hiding extensions and the file is actually called user.js.txt or something like that.

If you see alert message but something else goes wrong when you put the code snippet to hide columns then check browser console log for any errors and copy them here.

ArnieO, 2021-02-15:
- No effect whatsoever. 
- `alert('test')` -> Still nothing.

It seems to be the right directory:
[image]

I don't think Windows hides file name extension (it is one of the first defaults I switch off for any PC I work on):
[image]

Browser console log
Not very familiar with that, so this is maybe not what you're asking for (Firefox browser):
[image]

qu1ck, 2021-02-15:
Yes, file name and path is correct.
What is your ibom version? Support for customization was added after v2.3 so you need to download latest from master branch.

ArnieO, 2021-02-15:
> What is your ibom version?

I downloaded and installed it from [Github](https://github.com/openscopeproject/InteractiveHtmlBom/releases) today, using this link: https://github.com/openscopeproject/InteractiveHtmlBom/releases/download/v2.3/InteractiveHtmlBom.zip

This is shown in the header of the generated web page:
[image]

There *is* a possibility that ibom was already installed on this PC but not visible in KiCad after a recent KiCad upgrade. (I have used ibom before but changed PC some months ago - so am no longer sure if it was on this or previous PC. But as I did not find ibom in the recommended install directory, I proceeded with download/install, and the icon appeared in PCBnew as expected.

Is there any way I can verify that it is indeed version 2.3 I am running, and not a previously installed one (maybe sitting in an other directory)?

qu1ck, 2021-02-15:
v2.3 is NOT what you want, you need never version which is not released yet. Get it from this link https://github.com/openscopeproject/InteractiveHtmlBom/archive/master.zip

Remember to delete old version before installing new one and restart kicad.

> This is shown in the header of the generated web page:

That shows pcb file date (either from metadata or filesystem update date)

ArnieO, 2021-02-15:
That's more like it! 👍 
Sorry I missed the information regarding version, and for bothering you. It works as described now!

**Thank you for this great tool and for helping me so promptly today!**

qu1ck, 2021-06-06:
FYI, dynamic hiding and reordering of columns was just merged

[image]

joao-v-gomes, 2021-08-03:
Hi guys!

I've been using the awsome tool for a few days and now i'm trying to apply the solution that qu1ck wrote on #88 but i'm having the same problem ArnieO had. I think the _user.js_ file is in the right place but i'm not seeing the _alert('test')_. See the print below:
[image]

I'm using the files from the .zip file that qu1ck suggested to ArnieO and updated the installation.

Any ideas?

Thanks!

qu1ck, 2021-08-03:
Do you get a bom at all? Can you grep the output file for `alert('test')` ? Some browsers may hide js alerts.

Also it's better to put plugin in home directory `~/.kicad/scripting/plugins` as wiki suggests.

joao-v-gomes, 2021-08-03:
Hi qu1ck! Thanks for the reply!

I did some more testing and changed the plugin installation location. Using the same as the wiki.

On my Debian 9, I'm using Kicad 5.0 and I get the following results:

- Standard BOM options work perfectly. Checkboxes, highlights, filters, etc.
- Custom _user.js_ don't work. 

On my Windows 10, i'm using Kicad 5.1 and get this results:

- Standard BOM options work perfectly. Checkboxes, highlights, filters, etc.
- Custom _user.js_ work. 

So, _maybe_ it's a issue with my KiCAD version or my old Debian.

I'm using the same files on both OS, but i've also realized that, on my Debian, I can't see the icon for the dynamic hiding and reordering of columns and in Windows i can see it. Check the print below of the Debian generated BOM:

[image]

It's not a problem for me to do my work in Windows. So, i guess this solves the problem. If you need me to test something else, let me know. 

By the way, the code on #88 doesen't really solves my problem. What i really need is something like this: use a checkbox to choose what items would be highlighted on board. Do you recall if anybody has opened an issue about something like this?

Once again, thanks for the reply! This plugin is amazing! Awesome job, qu1ck!

qu1ck, 2021-08-03:
It sounds like on your debian install you've put a release version of the plugin instead of development version. Last released version doesn't have customization support, you need latest from master branch.

> By the way, the code on #88 doesn't really solves my problem. What i really need is something like this: use a checkbox to choose what items would be highlighted on the board. Do you recall if anybody has opened an issue about something like this?

This already exists, use `Mark when checked` toggle in page settings (only visible when there are checkboxes). Also only in dev version.

joao-v-gomes, 2021-08-04:
I'll do more tests in Debian today and get back to you. 

Yes! `Mark when checked` is exactly what i need. 

Thanks qui1ck!

joao-v-gomes, 2021-08-04:
Hi qu1ck!

I have found my problem. 

For some reason, the plugin files i've been using were not the ones at  `.kicad/scripting/plugin`, they were somewhere else. I keept the ones in the right folder and deleted the others. Now, everything works, including `user.js` configs.

Thanks!

qu1ck, 2021-10-11:
This is now implemented, please test

[image]

anshulj07, 2024-07-17:
Hey, I love this plugin but I have a question. How can I check the part number permanently? I have created a additional field using user.js but I assume part number wont come in user.js because that is a default field. 

Looking forward to you reply.

qu1ck, 2024-07-17:
I can hardly understand what you are asking, what it has to do with user.js or this 3 year old issue.

Please open a new issue and describe clearly what you want to achieve and what you have tried. If you are asking about how to make some extra field permanently enabled you can do it once and save current settings in a global config.

ArnieO, 2024-07-18:
> I can hardly understand what you are asking, what it has to do with user.js or this 3 year old issue.

And also a bit difficult to understand which comment this posting relates to! 😅