openscopeproject/InteractiveHtmlBom #168: Receiving AttributeError for TEXTE_PCB

bvkuik, 2020-07-10:
Hi, I have followed installation and the plugin is visible in KiCad. When running Generate BOM, I get the following error:

`Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Users/ME/Library/Preferences/kicad/5.99/scripting/plugins/InteractiveHtmlBom/ecad/kicad.py", line 523, in Run
    ibom.run_with_dialog(parser, config, logger)
  File "/Users/ME/Library/Preferences/kicad/5.99/scripting/plugins/InteractiveHtmlBom/core/ibom.py", line 328, in run_with_dialog
    main(parser, config, logger)
  File "/Users/ME/Library/Preferences/kicad/5.99/scripting/plugins/InteractiveHtmlBom/core/ibom.py", line 293, in main
    pcbdata, components = parser.parse()
  File "/Users/ME/Library/Preferences/kicad/5.99/scripting/plugins/InteractiveHtmlBom/ecad/kicad.py", line 469, in parse
    drawings, pcbnew.F_SilkS, pcbnew.B_SilkS),
  File "/Users/ME/Library/Preferences/kicad/5.99/scripting/plugins/InteractiveHtmlBom/ecad/kicad.py", line 201, in parse_drawings_on_layers
    drawing = self.parse_drawing(d[1])
  File "/Users/ME/Library/Preferences/kicad/5.99/scripting/plugins/InteractiveHtmlBom/ecad/kicad.py", line 168, in parse_drawing
    return self.parse_text(d)
  File "/Users/ME/Library/Preferences/kicad/5.99/scripting/plugins/InteractiveHtmlBom/ecad/kicad.py", line 159, in parse_text
    "thickness": d.GetThickness() * 1e-6,
AttributeError: 'TEXTE_PCB' object has no attribute 'GetThickness'`

This is on macOS 10.15.5 with KiCad nightly build of v6: reported as 5.99.0-2063-g36e7744ac.

qu1ck, 2020-07-10:
This is fixed already, update your plugin to latest from master branch.

closed 2020-07-10 by qu1ck

bvkuik, 2020-07-10:
Thanks, I downloaded the latest release from the page but I didn't check when that was updated. I'll try installing from master.