openscopeproject/InteractiveHtmlBom #170: Custom shape pads in a footprint cause the plugin to exit with an error

Diametrix, 2020-07-27:
I've been using your plugin with nightly builds and it seems to work on most of the designs.  However the designs that use custom shape pads (polygons) make the plugin to exit with error.  The one I get is: "AttributeError : 'D_PAD' object has no attribute 'GetCustomShapeAsPoligon'  

Thank you for the plugin by the way!  It is fantastic!

qu1ck, 2020-07-28:
Thanks for the report, I found a workaround but also filed a kicad issue https://gitlab.com/kicad/code/kicad/-/issues/4999
Will push a fix as soon as that is resolved one way or the other.

qu1ck, 2020-08-10:
I restored old api compatibility in nightly so this should be fixed once nightly builds pick it up.
Also fixed multilayer copper zones which was added in nightly, 98a73051d242a44d0958faffc65defbe0ce927f8.

closed 2020-08-10 by qu1ck