openscopeproject/InteractiveHtmlBom #172: Some BGA packages have two "first" pins highlighted.

whinis, 2020-08-12:
I noticed on one of my boards that both A1 and  P1 pins were high lighted as pin 1 in the view. The package is self made and is a BGA_180_12x12x0.8 package. with multiple missing pins in the center.

qu1ck, 2020-08-12:
There is always some guesswork involved in determining first pad as that information is not in kicad data model.

Right now any pad with name '1', 'A', 'A1', 'P1', 'PAD1' will be considered first. If there is no such pad then one with lexicographically smallest name will be first. And it's possible that there are multiple first pins because pads can be duplicated.

If you have suggestions for better algorithm, I'm open.

whinis, 2020-08-12:
Do you have an example of multiple first pins? that seems counter intuitive

qu1ck, 2020-08-12:
As an example this usb micro footprint has 4 pins named '6'

[image]

Same way there can be a footprint with few pins named 'A'.

whinis, 2020-08-12:
Ah yes, seems like that would be for abnormal parts however where the "first" pin is not obvious due the part being asymetrical. I would suggest you pin 1 pin and select the one with the lowest alphanumeric value. This way A1 is always above P1 or B1 but not 1. Likely more edge cases but less likely to get a scenario where its ambiguous which pin is actually first.

commit 1d379509 (2020-08-15): Improve pin1 detection

If there are multiple pads that look like common first pad names only
the lexicographically smallest one will be considered first.
If no pads have common first pad name then all pads with
lexicographically smallest name will be considered first (previously
only one random pad with such name was chosen).

Fixes #172
closed 2020-08-15 by qu1ck