openscopeproject/InteractiveHtmlBom #176: mark placed parts also on assembly view

nopeppermint, 2020-08-26:
If I manually check the placed parts I always mark the placed parts on my bom AND also on my assembly view with a light pen.

In InteractiveHtmlBom the marking inside the bom is implemented done with the cross in the checkbox but the assembly view is always the same, independent of the checkbox state.

For me that's a missing feature.

@qu1ck if you could give me a hint where to start I might try to implement this on my own

current:
C1 is selected and C2 is already placed
[image]

new (mockup):
C1 is selected and C2 is already placed (marked with green overlay on assembly view)
[image]

I am not sure if a green overlay is the best color, maybe someone else has a better idea?

qu1ck, 2020-08-26:
I like this idea but I think UI needs to be considered, this should be an option and probably somehow combined with "Darken when checked" as they are very similar.

If you want to look how it may be done take a look at getStoredCheckboxRefs() in ibom.js which will return list of component indexes that are currently marked for the given checkbox and drawHighlights()/drawModules() in render.js for how highlighting is done. You could just do second pass over all modules and only draw ones with index that is included in corresponding checkbox refset and draw them with transparent green.

nopeppermint, 2020-08-27:
> UI needs to be considered, this should be an option and probably somehow combined with "Darken when checked" as they are very similar.

I agree

And thanks for the hints where to start

commit 7f2f4f34 (2021-07-24): Darken when checked highlights footprints on render

Based on work by @unsynchronized, see
https://github.com/openscopeproject/InteractiveHtmlBom/pull/203

Fixes #176
Fixes #203
closed 2021-07-24 by qu1ck