openscopeproject/InteractiveHtmlBom #178: Question: interface the plugin with an external server

mngr0, 2020-09-02:
-----What I would like to do:
I would like to have the component rails managed by something, with a led for each rail. When I hoover the mouse over a component (or list of components of the same type) I would like to see the corresponding led to switch on.

-----What I THINK I need to do:
Since InteractiveHtmlBomDemo runs on client-side my something that manage the rails should be a server, to which the client make requests, with a fixed ip or something like that

-----And finally: ASSUMING that I think right, where should I put my hands? (on the code, I mean)
Where is the code that manage the highlighting of the component, when the mouse is over them?
Is it possible to make a request to a third-party server without suffering too much?

-----Thank you for your time and your attention, I am really a noob when we talk about web-clients-and-servers

qu1ck, 2020-09-02:
Good news is you don't have to modify plugin code at all, this use case is covered by the customization system already. There is a hook specifically for highlight event, see https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Customization#custom-javascript

All you have to do is define a callback function and register it for highlight event, ibom will call it whenever highlight changes and pass it a list of highlighted footprints. From there you can make a http call to another server or do anything your browser's js engine allows you to do.

mngr0, 2020-09-04:
Thank you very much, I am now working on this.
I have also learned that I must pay attention to the "wiki" button on github :)

You can manage this issue as you wish now

qu1ck, 2020-09-04:
Cool, I'll close this but if you have more questions feel free to reopen.

closed 2020-09-04 by qu1ck

mngr0, 2020-09-08:
Hello, I have found one more question:

My "rail manager" knows the component code and the comopnent package if we talk about simple resistors.
In the event that I can log there is only the name of the component (es. R3 or U2)

How can I get the value and/or the package?