openscopeproject/InteractiveHtmlBom #179: Saving the HTML with the checkboxes checked

JackTheEngineer, 2020-09-03:
Hello !

Thank you for creating this awesome plugin ! It is really great. 
Is it possible to save the checkbox state within the HTML file ? 
It would be really great to have this portable format to exchange information about what parts are missing / sourced.

Best regards

JackTheEngineer, 2020-09-03:
In case that it's possible, would it be possible to add an optional comment field, that gets saved too ?

qu1ck, 2020-09-04:
No, it's not possible to save state or comments in html as the whole bom table is generated dynamically every time you load the page. But you can share checkbox state by doing config export/import.

[image]

qu1ck, 2020-09-04:
An easy way to share comments would be to export bom table to a spreadsheet like google sheets or excel and have your comments there.
[image]
Copy button next to the filter field will put the bom table in cipboard from where you can just ctrl-v it into spreadsheet.

closed 2020-09-10 by qu1ck