openscopeproject/InteractiveHtmlBom #180: ERROR Please draw pcb outline on the edges layer

mr-miky, 2020-09-06:
I was using the nightly version of kicad aka 6.0.0.rc and this error appeared.
The strange thing is that if I loaded an old pcb version 5 the problem does not arise. I investigated a bit and discovered that the pcb outline is now composed of a rectangle and no longer of 4 segments.
Basically it happens that if you start the menu "Edit / Celeanup gfx" the outline of the pcb is optimized in rectangle.
The same problem happens if you draw the outline using the rectangle tool.

Theoretically `parse_draw_segment(self, d):`  should be taught how to deal with rectangles.
[test.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/5179731/test.zip)

qu1ck, 2020-09-07:
Thanks for reporting and providing test project. This should be fixed now, get the latest from master branch.

closed 2020-09-07 by qu1ck

mngr0, 2020-09-08:
Hello, I have found one more question:

My "rail manager" knows the component code and the comopnent package if we talk about simple resistors.
In the event that I can log there is only the name of the component (es. R3 or U2)

How can I get the value and/or the package?

commit f546c922 (2020-09-07): Add support for S_RECT drawsegment

Fixes #180