openscopeproject/InteractiveHtmlBom #181: sending value and package to external server

mngr0, 2020-09-08:
Hello, I have found one more question, form issue #178 , but I am not able to reopen it:

My "rail manager" knows the component code and the comopnent package if we talk about simple resistors.
In the event there is only the name of the component (es. R3 or U2).
How can I get the value and/or the package?

qu1ck, 2020-09-08:
All component data is available in the pcbdata.bom.both table. Look at dataformat doc.

For efficient lookup you can build a reverse index to quickly find bom row by component id:
```js
var component_lookup = [] //put it in global/window namespace to not have to pass it around
for (var i in pcbdata.bom.both) {
  for (var ref of pcbdata.bom.both[i][3]) {
    component_lookup[ref[1]] = i;
  }
}
```

After doing this once you can quickly get component data with
```js
var value = pcbdata.bom.both[component_lookup[component_id]][1]
var footprint = pcbdata.bom.both[component_lookup[component_id]][2]
```

mngr0, 2020-09-15:
Thank you very much for your time and for the answer,

I implemented that, but I was thinking: can I found a way to search by ID?
Such ID could be a manufacturer ID, or a custom ID.

In EEschema I can add extra fields to a component. Those extra data can be exported in the net-list or in the EESchema-BOM, but for what I know PcbNew does not pick them up.

I would like to make InteractiveHtmlBom to know such ID, how hard do you think it can be?

Kind regards
mngr

qu1ck, 2020-09-15:
This is already possible. After adding your custom field in eeschema export the netlist file and ibom will pick it up. On the extra fields tab in the ibom dialog make sure to tick the fields you want included in the bom and they will be available in pcbdata.

closed 2020-09-25 by qu1ck