openscopeproject/InteractiveHtmlBom #182: PCBnew 5.99 virtual components troubles

mr-miky, 2020-09-10:
I have the impression that there is some problem with the nightly version of pcbnew because the generated bom lacks all components except those that should not be mounted, I would expect the opposite.
I found that if I remove "Blacklist virtual components" they reappear.

mr-miky, 2020-09-10:
Another problem is that if DNP field name is activated, the component with dnf in the specified field still appears in the bom

qu1ck, 2020-09-10:
> generated bom lacks all components except those that should not be mounted, I would expect the opposite.

Yeah nightly added more attributes recently and they are incompatible with previous attributes. Need to fix.

> Another problem is that if DNP field name is activated, the component with dnf in the specified field still appears in the bom

That I can't reproduce. Don't forget to reexport netlist after setting dnp field.

closed 2020-09-10 by qu1ck

mr-miky, 2020-09-10:
Forgive me, I'm under stress and I forgot to regenerate the bom on eeschema.
In any case, among the various options of the component on EESchema there is an attribute "Exclude from bill of material", if I activate it the component still appears in the interactive html bom list, if I deactivate it it respects the DNP
Mysteries ...
P.S. I'm still using the version you fixed a few days ago

qu1ck, 2020-09-10:
> In any case, among the various options of the component on EESchema there is an attribute "Exclude from bill of material", if I activate it the component still appears in the interactive html bom list, if I deactivate it it respects the DNP

With the fix I just committed it should work correctly now.

set-soft, 2020-09-10:
> Yeah nightly added more attributes recently and they are incompatible with previous attributes. Need to fix.

I was expecting that and already commented about the impact in a thread about variants in the bug tracking.
I can't understand why they changed all the bits instead of let the two LSBs remain the same.

qu1ck, 2020-09-10:
> I can't understand why they changed all the bits instead of let the two LSBs remain the same.

It wasn't a bitmask at all and it would work to keep the meaning of those values. Now they work correctly in nightly and new attributes will just use unoccupied bits.

mr-miky, 2020-09-10:
I have now updated interactive html bom but everything marked "Exclude from bill of material" even if DNP appears in the bom.
PCBNew version hits something, I have tonight's nightly ...

qu1ck, 2020-09-10:
"Exclude from bill of material" checkbox in eeschema/pcbnew (they are synchronized when you do update from schematic) only works if you also tick "blacklist virtual" in ibom settings.
DNP field works regardless of those checkboxes but needs up to date netlist/xml file.
Also after updating the plugin restart kicad completely by closing all windows.

If you still have issues can you attach your project so I can debug? I can not reproduce in my tests.

mr-miky, 2020-09-10:
` If you still have issues can you attach your project so I can debug? I can not reproduce in my tests.`

I try tonight at home with some personal projects that have no disclosure problems.

commit d5dcef4c (2020-09-10): Use MOD_EXCLUDE_FROM_BOM attribute from nightly

It's equivalent of v5's virtual attribute

Fixes #182