openscopeproject/InteractiveHtmlBom #184: Any plans to make iBOM compatible with v5.99 nightlies?

WhiteChairFromIkea, 2020-09-15:


qu1ck, 2020-09-15:
I make nightly compatibility fixes all the time but you have to use the latest version of the plugin from master branch, not release version.
Is there any thing in particular that doesn't work?

WhiteChairFromIkea, 2020-09-15:
Yes, iBom failed on v5.99 (todays nightly) with "realworld" pcb, so I thought iBom is not chasing v5.99 updates, used it only with v5.1 branch, so asked here :)  
Also checked now, that iBom + v5.99 run successfully for simple PCB with several footprints and single zone, so everything is setup correctly. My current pcb is private and not so important to support iBOM, so I will close this.

In case you are willing to dig in, here is an error list which is thrown:
[image]

closed 2020-09-15 by WhiteChairFromIkea

qu1ck, 2020-09-15:
That's a bug in kicad, not in my plugin. It would actually help if you could share your pcb file. You can remove all footprints and traces if it's sensitive project. Judging by the error you have some uncommon object on the pcb that the kicad's python api doesn't process correctly, I would like to know what it is to report on their bugtracker.

WhiteChairFromIkea, 2020-09-16:
Solved:  
Removing all dimensions from the PCB fixed the issue (tried two of my boards).

qu1ck, 2020-09-16:
Thanks, I'll try to reproduce this and offer a fix upstream or at least report the bug with a repro steps.

qu1ck, 2020-09-18:
For posterity, I fixed this upstream in https://gitlab.com/kicad/code/kicad/-/merge_requests/419