openscopeproject/InteractiveHtmlBom #185: AttributeError when clicking "Generate BOM" button

PaulPillau, 2020-09-16:
When I click "Generate BOM" in the GUI, Python throws this exception:

[image]

This stack trace comes from an almost empty test project that only contains one connector and edge cuts.
I first tried it with my current project but got the same error.

Note that I'm using a portable version of KiCAD 5.1.6 which is available here:
[https://github.com/2wayne/KiCadPortable](url)
I installed the plugin in "C:\KiCadPortable\App\KiCad\share\kicad\scripting\plugins\"

qu1ck, 2020-09-16:
That's an issue with the installation. To verify it you can try running this in scripting console:

```python
import pcbnew
b = pcbnew.GetBoard()
m = list(b.GetModules())[0]
mcopy = pcbnew.MODULE(m)
```
The last line should fail judging by the stack trace from the screenshot. This indicates that pcbnew API is broken, likely not all files were properly updated in the portable kicad install.

PaulPillau, 2020-09-16:
That last line is indeed the problem.
I installed 5.1.6 normally and it works perfectly now.

Thanks for the quick reply!

closed 2020-09-16 by PaulPillau