openscopeproject/InteractiveHtmlBom #190: Unable to generate bom. PcbNew 5.99

mr-miky, 2020-10-05:
Every time I try to make the bom generate the error shown in the attached screenshot appears.

[image]

Application: Pcbnew

Version: 5.99.0-unknown-r20523-27f3ac98, debug build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.69.1 OpenSSL/1.1.1g-fips zlib/1.2.11 brotli/1.0.7 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh/0.9.5/openssl/zlib nghttp2/1.41.0

Platform: Linux 5.8.12-200.fc32.x86_64 x86_64, 64 bit, Little endian, wxGTK, mate, x11

Build Info:
	Date: Oct  5 2020 18:56:50
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.69.0
	OCC: 7.4.0
	Curl: 7.69.1
	ngspice: 32
	Compiler: GCC 10.2.1 with C++ ABI 1014

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=ON
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
	KICAD_SCRIPTING_ACTION_MENU=ON
	BUILD_GITHUB_PLUGIN=ON
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
	KICAD_STDLIB_DEBUG=OFF
	KICAD_STDLIB_LIGHT_DEBUG=OFF
	KICAD_SANITIZE=OFF
`
`

qu1ck, 2020-10-05:
[This](https://gitlab.com/kicad/code/kicad/commit/37906511f585fd635a0d8a233472c79406407dee) broke a lot of plugins.
Sigh...

qu1ck, 2020-10-13:
API fixed in https://gitlab.com/kicad/code/kicad/-/merge_requests/467

Update both the nightly and the plugin.

commit df2f0dd1 (2020-10-13): Fix nightly compatibility

Fixes #190
Fixes #189
Fixes #187
closed 2020-10-13 by qu1ck