openscopeproject/InteractiveHtmlBom #193: Still seeing some silkscreen issues on Eagle imports

kasbah, 2020-10-08:
Hey, following on from #183. We are still seeing some issues with silkscreen items, graphical ones, that look fine in KiCad but are scaled or transposed in IBOM. This is with the same [kicad_pcb files](https://github.com/openscopeproject/InteractiveHtmlBom/files/5210344/kicad_pcbs.zip). I am not sure what the culprit is yet but you can see it e.g. on the flame: 

[image]

http://eagle-ibom.preview.kitspace.org/interactive_bom/?github.com/pyr0ball/pyr0piezo/Pyr0-Piezo%20Z-Probe%20Business%20Card%20DIY%20Module

In KiCad:

[image]

```kicad_pcb
  (module Pyr0-Piezo-v2-DIY-Business-Card:LOGO-PYR0BALL-KEEPOUT-20MM (layer Top) (tedit 0) (tstamp 5F4FABC3)
    (at 168.1861 87.8586)
    (fp_text reference U$1 (at 0 0) (layer F.SilkS) hide
      (effects (font (size 1.27 1.27) (thickness 0.15)))
    )
    (fp_text value "" (at 0 0) (layer F.SilkS) hide
      (effects (font (size 1.27 1.27) (thickness 0.15)))
    )
    (fp_poly (pts (xy -6.773831 0.172465) (xy -6.774159 0.133303) (xy -6.774159 0.094075) (xy -6.774159 0.026962)
    ...
   ))

```

or the text here:

[image]

http://eagle-ibom.preview.kitspace.org/interactive_bom/?github.com/joeycastillo/The-Open-Book/eBook-Wing

In KiCad:

[image]

```kicad_pcb
  (module OSO-BOWI-A1-05:OSO_WORDMARK_125X1300MIL_1000DPI (layer Top) (tedit 0) (tstamp 5F50014C)
    (at 153.8351 146.4056)
    (fp_text reference U$11 (at 0 0) (layer F.SilkS) hide
      (effects (font (size 1.27 1.27) (thickness 0.15)))
    )
    (fp_text value "" (at 0 0) (layer F.SilkS) hide
      (effects (font (size 1.27 1.27) (thickness 0.15)))
    )
    (fp_poly (pts (xy 8.1153 -1.5621) (xy 8.1407 -1.5621) (xy 8.1407 -1.5875) (xy 8.1153 -1.5875)) (layer F.SilkS) (width 0))
    (fp_poly (pts (xy 7.3025 -1.5621) (xy 7.6073 -1.5621) (xy 7.6073 -1.5875) (xy 7.3025 -1.5875)) (layer F.SilkS) (width 0))
    ...
    )
```

qu1ck, 2020-10-08:
I can't reproduce, when I open the files locally they generate correct boms:

[image]
[image]

I remember there was a bug in pcbnew that caused python API to return incorrect offset values after eagle import. But if you closed and reopened the board, then it would be fine. I fixed some of those issues but probably not all.

I'm not sure how your script handles eagle imports but try doing the import, save the file somewhere temporarily and reopen it.

kasbah, 2020-10-08:
Thanks @quick. I was thinking I should try and reproduce in mainline IBOM first but was too lazy. Serves me right :D

closed 2020-10-15 by qu1ck