openscopeproject/InteractiveHtmlBom #195: 5.99 nightly PCB outline error

mgiacomelli, 2020-11-07:
Following the comment in issue 190, I updated to the current KiCAD nightly (5.99.0-6862-g9454f9df92) as well as the current IBOM git.  Unfortunately, hitting "Generate BOM" gives:

"Please draw pcb outline on the edges layer on sheet or any module before generating BOM."

I do have a PCB outline, and prior to updating KiCAD the files worked correctly.  Did I do something wrong or did KiCAD break again?

qu1ck, 2020-11-07:
I just compiled kicad from master branch and can't reproduce the issue but I test on a board with simple square outline.

Maybe yours has outline consisting only of arcs or something? Could you attach your board for me to test? Feel free to remove everything from it aside from the outline and maybe one footprint.

mgiacomelli, 2020-11-07:
My board is circular...

Here is a different open source board with the same issue:  https://github.com/mgiacomelli/sipm/tree/master/Improved%20Design/KiCAD

[image]

qu1ck, 2020-11-07:
I was able to generate bom with the board you linked just fine.

Can you confirm that your ibom version is on df2f0dd ?

mgiacomelli, 2020-11-07:
I'm sorry, there was actually an older copy also on the kicad plugins search path, and that was being found before the current git version of the plugin.  Deleting the old one fixes the problem.

closed 2020-11-07 by mgiacomelli