openscopeproject/InteractiveHtmlBom #196: Please Draw PCB Outline

greenscreenflicker, 2020-11-12:
Hello,
i get the following issue when generating the bom. It's an PCB array.
It has edge outlines.
[image]

Best regards,
Michael

greenscreenflicker, 2020-11-12:
Kicad version information
Application: Pcbnew

Version: (5.99.0-6849-g69c1171320), release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.71.0 OpenSSL/1.1.1g (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.41.0

Platform: Windows 10 (build 10586), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Nov  6 2020 11:04:13
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8)
	Boost: 1.73.0
	OCE: 6.9.1
	Curl: 7.71.0
	ngspice: 32
	Compiler: GCC 10.2.0 with C++ ABI 1014

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=OFF
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
	KICAD_SCRIPTING_ACTION_MENU=ON
	KICAD_USE_OCE=ON
	KICAD_SPICE=ON

greenscreenflicker, 2020-11-12:
Happens with empty pcb also
[image]

qu1ck, 2020-11-13:
This is fixed in latest revision, update your plugin from master.

closed 2020-11-13 by qu1ck

greenscreenflicker, 2020-11-16:
@qu1ck I used for testing the following commit. Please advice, if I missed an update.
[image]

qu1ck, 2020-11-16:
No, that is the latest commit. 
You need to make sure that it was actually running, i.e. you don't have another older ibom folder in your plugins directory and you restarted kicad fully after updating the plugin.

greenscreenflicker, 2020-11-18:
I confirmed that i am using the latest version. Error is still there.

qu1ck, 2020-11-19:
I cannot reproduce your issue. I get this for an empty board on a recent nightly. You either don't have most recent nightly or most recent plugin.

[image]

The path in tortoisegit window suggests that you have the plugin checked out outside of kicad plugin directory. Make sure you have your links setup correctly or that you have copied the folder over recently. When updating the plugin by copying files always completely remove old plugin directory. If you have a filesystem symlink setup as described in installation instructions then simply doing a git checkout is fine.

Btw I just pushed another commit that fixes nightly compatibility. You need to update plugin to latest.