openscopeproject/InteractiveHtmlBom #197: Error with current KiCad nightly

zonque, 2020-11-19:
When running with the latest KiCad nightly, the generation of files fails like this:

[image]
(sorry for the image post, I can't copy the text from that dialog)

qu1ck, 2020-11-19:
Already fixed in 43e5034, please update your plugin.

closed 2020-11-19 by qu1ck

zonque, 2020-11-19:
Oops, sorry. I did that of course, but neither "Refresh plugins" nor restarting pcbnew sufficed to pick up the change. I had to restart the project manager as well to make it effective.

qu1ck, 2020-11-19:
See point 2 here:
https://github.com/openscopeproject/InteractiveHtmlBom/wiki/FAQ#q-i-updated-plugin-to-a-recent-version-and-nothing-changed-or-plugin-behaves-incorrectly

zonque, 2020-11-19:
:bow: