openscopeproject/InteractiveHtmlBom #198: Subscripts in Text elements not rendering properly

arturo182, 2020-11-19:
KiCad nightly added support for having sub- and superscripts in schematics and in the PCB Text Elements, when you type Foo_{bar} you get "bar" as subscript, and for Foo^{bar} you get it as superscript.

Here's an example:
[image]

[image]

These are currently not rendering properly in the BOM:

[image]

KiCad version: 5.99.0-6923-g2a3e6ee1b5
InteractiveHtmlBom version: 2.3-33-g43e5

Not really a showstopper, but still a bug 🙂

qu1ck, 2020-11-19:
Yeah I saw these added a while ago and seeing how subdued the effect is I though likely nobody would use it in pcbs, only on schematics. Of course, I was wrong.

KiCad's text rendering code is becoming more and more convoluted and what's worse, ever so slightly inconsistent between versions. Even things like overbar rendering, interline distance changed a bit in v6. I'll try to look for a way to get text as polyline object from kicad itself instead of trying to replicate it's rendering code in ibom. It might blow up the generated page size a bit but it's probably worth it.

qu1ck, 2020-11-21:
Relevant MR in kicad https://gitlab.com/kicad/code/kicad/-/merge_requests/530
If that is accepted I have a patch ready that will make ibom's own implementation of text rendering obsolete.

arturo182, 2020-11-21:
Awesome!

qu1ck, 2020-11-23:
Wait a day for kicad's nightly to pick up my change and then update to latest.

commit 7d78c119 (2020-11-23): Use kicad's own text rendering when possible

Transform kicad text objects into segments list using native rendering
method and then build an svgpath from it for the web renderer.

Fixes #198
closed 2020-11-23 by qu1ck