openscopeproject/InteractiveHtmlBom #201: Nigthly Issue: Board has no attibute get modules

greenscreenflicker, 2020-11-24:
@qu1ck I have the following issue:
[image]

Your are also newest:
[image]

I use newest nightly (Version 23.11.2020)

Application: Pcbnew

Version: (5.99.0-7170-gc67c13bc1d), release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.71.0 OpenSSL/1.1.1g (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.41.0

Platform: Windows 10 (build 10586), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Nov 23 2020 23:22:00
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8)
	Boost: 1.73.0
	OCE: 6.9.1
	Curl: 7.71.0
	ngspice: 32
	Compiler: GCC 10.2.0 with C++ ABI 1014

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=OFF
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
	KICAD_SCRIPTING_ACTION_MENU=ON
	KICAD_USE_OCE=ON
	KICAD_SPICE=ON

qu1ck, 2020-11-24:
This is fixed in master already.

Stack trace in your first screenshot shows outdated code. Did you restart kicad after updating the plugin?

greenscreenflicker, 2020-11-24:
@qu1ck I have the impression, there must be ghosts somewhere.
If I update to the last revision (as shown above) it produces the same error.
If I delete the folder, restart kicad, the tool is gone too. 
If I copy then the folder, refreh the folder, the tool produces the same error. (And yes, I use the following commit: 7d78c1194e7d73a0058633ba9e9cd0f63427235d )

qu1ck, 2020-11-24:
Without seeing your setup the only thing I can suggest is do a completely clean install.
Close kicad, delete the plugin folder from kicad scripting path, delete your local git checkout.
Then do a clean git checkout straight in `%APPDATA%\kicad\scripting\plugins`

closed 2020-11-30 by qu1ck

greenscreenflicker, 2020-12-02:
Checking the git repository out again solved the problem. Thanks for your assistance. Highly apprechiated.