openscopeproject/InteractiveHtmlBom #202: Specify manual sort order for specific components

wohali, 2020-11-24:
I was wondering if specific components could be added to the Component sort order dialogue. So, for instance, I want C1-C5 to be populated much later than the rest of the capacitors, because they are big and get in the way of D, U, J, etc.

This would require that matching on "C" exclude any explicitly entered components like C1, C2, etc.

qu1ck, 2020-11-24:
No, this is a very narrow use case.
What you can do instead is add an extra field "population order" or whatever you want to call it and then sort the table by that field.

wohali, 2020-11-29:
How is this a narrow use case? What if half of your capacitors are SMD, and the other half are electrolytics? You might want to populate the SMD ones, then the SMD ICs, and finish off with the through-hole parts later.

How does the extra field ordering work?

qu1ck, 2020-11-29:
It's a narrow use case because it falls within "custom sorting" which is already achievable.

Look at this demo for an example of extra field https://openscopeproject.org/InteractiveHtmlBomDemo/html/Keithley1950.html

It has "comment" column, you can click on the header to change sorting order.

Similarly you can add your own extra field and sort by that.

wohali, 2020-12-01:
OK, so I'd need #124 to make this useful for my customers. Thanks.

closed 2020-12-01 by wohali

qu1ck, 2020-12-01:
In the short term you can achieve the desired default sorting behavior with a bit of custom js.