openscopeproject/InteractiveHtmlBom #204: vertical divider sticks to mouse

nopeppermint, 2020-11-30:
I just tested current master #7d78c11 and there under Win10 x64 with Firefox 83.0 64bit (and also with Chrome v.  86.0.4240.198 x64) once clicked on the vertical devider it sticks to the mouse forever, only reload of the page helps.

This is a new issue, before (sry. don't know the version before)

[image]

qu1ck, 2020-11-30:
Can't reproduce. Here is a demo generated with latest kicad and latest ibom:
https://openscopeproject.org/InteractiveHtmlBomDemo/html/rex.html

It works fine for me.

What you are seeing is likely caused by some javascript errors. Try following
* If you have any custom `user.js` added to ibom, see if removing it helps.
* Do a clean install of ibom plugin
* Open developer tools in chrome and see if there are any errors in the console. Copy them here.

qu1ck, 2020-11-30:
One more thing to try:
* See if the issue happens only with a particular board. Try generating bom for one of kicad demo projects and see if issue reproduces.

nopeppermint, 2020-11-30:
> If you have any custom user.js added to ibom, see if removing it helps

I have not

> Do a clean install of ibom plugin

I just did

> Open developer tools in chrome and see if there are any errors in the console. Copy them here.

```ibom.html:1957 Uncaught ReferenceError: allcanvas is not defined
    at resizeAll (ibom.html:1957)
ibom.html:1407 Uncaught TypeError: text.pos is not iterable (cannot read property undefined)
    at drawText (ibom.html:1407)
    at drawBgLayer (ibom.html:1741)
    at drawBackground (ibom.html:1852)
    at redrawCanvas (ibom.html:1941)
    at resizeCanvas (ibom.html:1953)
    at resizeAll (ibom.html:1957)
    at changeCanvasLayout (ibom.html:2974)
    at changeBomLayout (ibom.html:3097)
    at window.onload (ibom.html:3345)
ibom.html:1407 Uncaught TypeError: text.pos is not iterable (cannot read property undefined)
    at drawText (ibom.html:1407)
    at drawBgLayer (ibom.html:1741)
    at drawBackground (ibom.html:1852)
    at redrawCanvas (ibom.html:1941)
    at resizeCanvas (ibom.html:1953)
    at resizeAll (ibom.html:1957)
```

nopeppermint, 2020-11-30:
> See if the issue happens only with a particular board. Try generating bom for one of kicad demo projects and see if issue reproduces.

with this Project https://github.com/c4puter/motherboard it does not happen..

nopeppermint, 2020-11-30:
but it happens with another from Altium imported project:

https://github.com/issus/USBLiPoBatteryCharger

can you reprodoce it with this project?

nopeppermint, 2020-11-30:
my output:
[ibom.html.txt](https://github.com/openscopeproject/InteractiveHtmlBom/files/5615712/ibom.html.txt)

Errors on Chrome:
[chrome.log](https://github.com/openscopeproject/InteractiveHtmlBom/files/5615709/chrome.log)

it's the same for https://github.com/openscopeproject/InteractiveHtmlBom/issues/205

qu1ck, 2020-11-30:
Thanks, I was able to reproduce. Should be fixed now.

closed 2020-11-30 by qu1ck

nopeppermint, 2020-11-30:
indeed, fixed with https://github.com/openscopeproject/InteractiveHtmlBom/commit/0ecd3b7ef13ebe1b2cdb833930a4b48f8a3def59 thanks

commit 0ecd3b7e (2020-11-30): Fix handling of empty text objects in nightly

If text has no visible text (only whitespace) then svgpath will be
empty.

Fixes #204
Fixes #205