openscopeproject/InteractiveHtmlBom #205: bom list on startup not displayed, even with empty filter

nopeppermint, 2020-11-30:
I just tested current master #7d78c11 and there under Win10 x64 with Firefox 83.0 64bit (and also with Chrome v. 86.0.4240.198 x64) the bom list ist empty on start, even when there is no filter applied.

If you write something and clean it then I expect the same result, but then the bom list is visible:

on startup:

[image]

after writing a filter and cleaning the filter:
[image]

This is a new issue, before (sry. don't know the version before) it was working.

qu1ck, 2020-11-30:
Can't reproduce this either. See my comment in #204 for potential solutions.

closed 2020-11-30 by qu1ck

nopeppermint, 2020-11-30:
indeed, fixed with https://github.com/openscopeproject/InteractiveHtmlBom/commit/0ecd3b7ef13ebe1b2cdb833930a4b48f8a3def59 thanks

commit 0ecd3b7e (2020-11-30): Fix handling of empty text objects in nightly

If text has no visible text (only whitespace) then svgpath will be
empty.

Fixes #204
Fixes #205