openscopeproject/InteractiveHtmlBom #206: Give unique URL to each specific BOM line

rusefillc, 2020-12-06:
it would be great to be able to link to specific BOM line or even specific referenced component to have it pre-selected when page opens

makerstorage, 2021-12-08:
Can we do that from the user.js?

get the url parameter (rowid) like url.html?rowid=5
and call the createRowHighlightHandler or any other function to highlight that row?

qu1ck, 2021-12-08:
Literally anything is theoretically possible to do in `user.js`.
But a good UI needs some more thought put into it. How to select rows for url change? Should multiple be selectable? Does bom mode need to be encoded in url too since it affects row numbers and bom contents? What about filters?

There's many questions I haven't thought through yet but I do like the idea. Of course you can do some minimum viable version in user.js, share it if you can make it work.

vanBassum, 2022-11-23:
Or maybe use the filters for this,
url.html?filter=R21