openscopeproject/InteractiveHtmlBom #209: Add option to *not* track mouse in parts list

PJMonty, 2020-12-07:
My standard workflow is to look at the footprint I want to solder on the PCB, then click on that same footprint in the preview window.  I then see what part I need, pull it from my parts bin, and then go to click "sourced" or "placed".  It's at this point that the nice, highlighted line in the parts list suddenly starts moving as my mouse is tracked and whatever line it's over becomes highlighted instead.

I'd like to suggest an option to disable live tracking the mouse in the parts list so that I can simply click in either the parts list or the preview window and have whatever I clicked highlight and stay hightlighted until I click something else.

qu1ck, 2020-12-07:
https://github.com/openscopeproject/InteractiveHtmlBom/issues/88#issuecomment-650539158

PJMonty, 2020-12-09:
Wow.  Well, that certainly works!  

It would still be nice to have this as a GUI option, but until then this is very much appreciated!

closed 2020-12-09 by PJMonty