openscopeproject/InteractiveHtmlBom #211: Request  to be able to add partlinks

bendelathouwer, 2020-12-11:
Is it possible to have a field in the html page so we can add the links to the part's mouser/farnell/ etc page

With best regards Ben

dreamcat4, 2020-12-12:
how about octopart? Does octopart have an http api we can use for that?

qu1ck, 2020-12-12:
You can use extra fields to put any data you want into the bom, including links. They will be displayed as simple text though, if you want clickable links you will have to write some custom js (see customization in wiki).

bendelathouwer, 2020-12-16:
> how about octopart? Does octopart have an http api we can use for that?

I'd like a digikey/mouser etc link easyer wen ordering tbh but octopart is a start

bendelathouwer, 2020-12-16:
> You can use extra fields to put any data you want into the bom, including links. They will be displayed as simple text though, if you want clickable links you will have to write some custom js (see customization in wiki).

So there is no easy way of adding part links to the bom? so I have to edit the htm page?

qu1ck, 2020-12-16:
You don't have to edit html page to add links. Just add custom fields to parts in eeschema.

closed 2020-12-18 by bendelathouwer