openscopeproject/InteractiveHtmlBom #215: Inconsistent Pin 1 designation (?) on WQFN footprint

mitchd, 2020-12-30:
I have a row of three WQFN-16 packages, and it looks like the 'symbology' for Pin 1 is incorrect. Note the silk around the parts designating Pin 1 as the southeast corner (this is the correct location). The grey body and white dot -- overlaid where the body pad is -- don't quite follow the part's orientation:

[image]

The part centroid and rotation data are all correct when exported from pcbnew:

```
U1102 | TLA2518 | WQFN-16-1EP_3.15x3.15mm_P0.5mm_EP1.7x1.7mm_ThermalVias_HS | 3.2392 | 5.502 | 0 | bottom
U1103 | TLA2518 | WQFN-16-1EP_3.15x3.15mm_P0.5mm_EP1.7x1.7mm_ThermalVias_HS | 3.9438 | 5.5014 | 0 | bottom
U1104 | TLA2518 | WQFN-16-1EP_3.15x3.15mm_P0.5mm_EP1.7x1.7mm_ThermalVias_HS | 4.6362 | 5.4996 | 0 | bottom
```

Footprint data attached for reference in case there's something I've done creating the footprint to cause this:

[WQFN-16-1EP_3.15x3.15mm_P0.5mm_EP1.7x1.7mm_ThermalVias_HS.kicad_mod.txt](https://github.com/openscopeproject/InteractiveHtmlBom/files/5754683/WQFN-16-1EP_3.15x3.15mm_P0.5mm_EP1.7x1.7mm_ThermalVias_HS.kicad_mod.txt)

qu1ck, 2020-12-30:
This has nothing to do with pin 1 designation (in fact you are not using built-in feature for pin 1 highlight because none of the passives have highlited pins).
The white dot on wqqfn package is the hole of the thermal via. It's only one instead of 2 and in various positions because of non-deterministic rendering order causing square smd pad 17 to sometimes draw over the through hole pads 17.

This is a known issue, up until now I considered it just cosmetic. But your case is quite confusing so I'll fix it.

mitchd, 2020-12-30:
Undestood, I figured it might be something along those lines hence the (?) there.

Other packages with exposed pads and thermal vias had a convenient white dot near where pin 1 is, which seemed oddly coincidental -- especially since highlighting Pin 1 is a discrete feature and uses a different method of indication.

commit 3440c144 (2020-12-30): Make pad hole rendering deterministic

Render holes after pads themselves so that overlapping pads
don't paint over the holes.

Fixes #215
closed 2020-12-30 by qu1ck