openscopeproject/InteractiveHtmlBom #217: "Exception on python action plugin code" for KiCad nightly

a7hybnj2, 2021-01-10:
I know that kicad is going through A LOT of changes currently for the 6.0 rollout but I thought I would post there here just in case this particular issue sticks around.
```
Traceback (most recent call last):

File "/Users//Library/Preferences/kicad/5.99/scripting/plugins/InteractiveHtmlBom/InteractiveHtmlBom/ecad/kicad.py", line 551, in Run
self.icon_file_name = os.path.join(icon_dir, 'icon.png')

File "/Users//Library/Preferences/kicad/5.99/scripting/plugins/InteractiveHtmlBom/InteractiveHtmlBom/core/ibom.py", line 339, in run_with_dialog
main(parser, config, logger)
```
```
Application: KiCad

Version: (5.99.0-8276-ge338d69cfa), release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.64.1 SecureTransport (LibreSSL/2.8.3) zlib/1.2.11 nghttp2/1.41.0

Platform: macOS Version 10.16 (Build 20C69), 64 bit, Little endian, wxMac

Build Info:
	Date: Jan  9 2021 04:26:40
	wxWidgets: 3.0.4 (wchar_t,STL containers,compatible with 2.8)
	Boost: 1.75.0
	OCC: 7.5.0
	Curl: 7.54.0
	ngspice: 31
	Compiler: Clang 10.0.1 with C++ ABI 1002

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=OFF
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
	KICAD_SCRIPTING_ACTION_MENU=ON
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

qu1ck, 2021-01-11:
I think you cut off the important part of the exception, the message itself. Can you post full log or screenshot?

a7hybnj2, 2021-01-11:
Well SOB. I went in to take a screenshot and there is no longer an error. I did update to the latest kicad nightly a few hours ago so I guess it was just a temporary problem.

closed 2021-01-11 by a7hybnj2