openscopeproject/InteractiveHtmlBom #225: FYI, no action needed: andreika-git-iBom with 3D view of the PCB

rusefillc, 2021-02-21:
Just FYI, maybe a useful feature proposal: 3D view of the PCB

at the moment it's merged into https://github.com/andreika-git/hellen-one/ and right now we are short on time to properly extract it as a separate tool or properly contribute it into the upstread.

https://rusefi.com/docs/ibom/hellen72b-ibom.html

[image]

Implementation detail: we reuse a modified version of iBom-template.html

qu1ck, 2021-02-23:
It's just a prerendered image drawn in the background. Nice, but I'd like to some day integrate [this](https://easyw.github.io/vrm360/).

ozel, 2021-03-30:
a 3D assembly view would be so nice and especially useful for electronic beginners! For this beginner-friendly project I've used scripts to generate resistor models with colour bands:
[image]
In such 'educational' use cases, the 2D iBOM webview is currently a bit too abstract for beginners and not really necessary of course for experienced tinkerers due to the simplicity of the board.

Together with @Kasbah we've briefly experimented with using the SVG board outline for generating a simple 3D board model in javascript. The component 3D model files could then be loaded separately as needed instead of working with one big STEP file.

qu1ck, 2023-08-30:
Cleaning up unactionable issues.

closed 2023-08-30 by qu1ck